ut-robotics/picr22-team-placeholder #9: MCU pin assignement

ReikoR, 2022-10-29:
[image]

Did you use STM32CubeIDE for creating the pin assignment? Current one does not make too much sense.

Some more information on how to configure timers in STM32CubeIDE is on the firmware page:
https://ut-robotics.github.io/picr22-home/basketball_robot_guide/electronics/firmware.html

Please ask if it's not clear how to do the timer configuration and pin assignment.

artursalumae, 2022-10-30:
Yep agreed, I had a wrong understanding in regards to how timers work in STM32. I will commit the fix soon.

ReikoR, 2022-11-06:
It's would be a bit better to have encoder channel A connected to timer channel 1 for all motors. Right now motor 1 encoder is connected this way, but other encoders are not. Otherwise it would need to be corrected in firmware or software.

[image]
[image]

closed 2022-11-12 by ReikoR