ut-robotics/picr22-team-placeholder #20: [Mechanics] CAM - 2D Contour issues

maidlam, 2022-11-05:
- Enable "Multiple depths" and set it according to the Passes tab settings here https://ut-robotics.github.io/picr22-home/basketball_robot_guide/mechanics/cam.html#_pocket
- Disable "Ramp" in the Linking tab.
- Change the "Cutting feedrate" to 1000 mm/min. Some other settings might change based on that.

ol4f, 2022-11-06:
might be ok now? didn't find the "cutting feedrate" from anywhere

maidlam, 2022-11-06:
For some reason, a 3 mm drill is now selected as the 2D Contour tool.

The "Cutting feedrate" is in the tool tab
[image]

Also, disable "Keep Tool Down" in the linking tab - it causes unwanted movements through the material
[image]

ol4f, 2022-11-06:
should be ok now

closed 2022-11-07 by maidlam