ut-robotics/picr22-team-placeholder #41: Why have 3 pieces in the motor mount?

Akustav, 2022-12-02:
Why do motor mounts consist of 3 components when one could do the same job?

[image]

ol4f, 2022-12-03:
Should give better stability. But for convenience I can make the main part bigger and discard the side mounts.

Added drillholes.

Akustav, 2022-12-04:
The previous robot had such a scheme to have sloped wheel angels. You do not have them so there should be no point in adding these additional pieces. Now there are some jointing issues:

[image]

ol4f, 2022-12-05:
Fixed.