ut-robotics/picr22-team-placeholder #49: How is the PC mounted

Akustav, 2022-12-02:
I see no mounting holes for the PC

[image]

ol4f, 2022-12-03:
added proper screwholes.

Akustav, 2022-12-04:
What do these holes attach to on the computer chassis? I dont remember mounting holes being there

ol4f, 2022-12-05:
I copied the screw placements from the old robot's NUC_plate. And the NUC is currently attached with that plate.

Akustav, 2022-12-05:
I think you want to use these holes for mounting and they do not line up with the current holes you have:

[image]

ol4f, 2022-12-05:
Found them, fixed. Waiting to save.