ut-robotics/picr22-team-placeholder #50: Blue lines

Akustav, 2022-12-02:
Why so many blue lines in the main chassis sketch? 

[image]

Also looks awfully like the previous years robot... Especially where the thrower socket fits. Dark grey areas are where the two bottom plates align. Weird...
[image]

ol4f, 2022-12-03:
Fixed blue lines.
The previous design was very inspirational :)

Akustav, 2022-12-04:
Okay... I will close this task later but FYI you can use the EQUALS constraint to define the circle size of one and then make them all equal. Less clutter on sketches that way:

[image]
###

closed 2022-12-05 by Akustav