ut-robotics/picr22-team-placeholder #52: No electronics in chassis

Akustav, 2022-12-04:
Consider where the electronics will sit in the chassis

ol4f, 2022-12-04:
We were thinking of putting them in middle of the motors, but drill the holes once we've half-assembled it, for optimal placing regarding wiring.

Akustav, 2022-12-05:
I mean you know the sizes, dont you? Why not let the machine do it. Also the step-up converter is rather large so it needs some consideration when fitting into the chassis

ol4f, 2022-12-05:
[image]
Added the holes to both sides, so the PCB and step-up could be put on either sides, depending on how the wiring is.

ol4f, 2022-12-05:
If you are wondering what the 4 holes in middle of the motors are, then that is also one possible PCB placement spot. The more choices, the merrier.

closed 2022-12-07 by maidlam